openWB/core #1975: Feature Request: Show alternative website in integrated display

MartinKist, 2024-10-25:
Just like Konni describes in [this forum thread](https://forum.openwb.de/viewtopic.php?t=8191), I would also like to use the evcc webinterface on the screens of my OpenWBs.

It would be nice if the website shown by the integrated display could be configured via the OpenWB webinterface.

If someone could point me in the right direction, I would try to add a pull request myself.

benderl, 2024-10-26:
Please have a look at my answer in the forum.
https://forum.openwb.de/viewtopic.php?p=100888#p100888

This is not a feature that we are working on, but feel free to open a PR.

Subito, 2025-03-21:
As far as I can see, Themes are disabled if the openWB is used as a secondary? In that case your suggestion would not work. Is that something that could be enabled?

benderl, 2025-03-21:
> As far as I can see, Themes are disabled if the openWB is used as a secondary?

This is not disabled. There is just no local theme yet for this use case. If you take a look at the display wrapper, this is already prepared but not activated due to the lack of local themes for secondaries

Subito, 2025-03-21:
Oh great! I'd be willing to test this and try to make a theme, could you point me at the display wrapper to get me started? Sorry, I'm not familiar with the codebase and that would speed things up a lot and would be greatly appreciated.

benderl, 2025-03-22:
https://github.com/openWB/core/blob/3d38c47f3559446b153c8d28ff3c4c6f73886461/web/display/processAllMqttMsg.js#L41

The display themes are located in `packages/modules/display_themes`.

gernotgernot, 2025-08-30:
would also love to have a simple solution for configuring another website url for the display when openwb is running as secondary