openWB/core #2095: Zeitladen: Phasenumschaltung wird nicht getriggert

MartinRinas, 2024-12-20:
Zeitladefenster aktiv, Ladevorgang startet mit der vorgegebenen Stromstärke, allerdings läuft timecheck in eine Fehler was vermutlich die Umschaltung von 1p auf 3p verhindert.

`2024-12-20 20:48:02,830 - {control.chargepoint.chargepoint:447} - {ERROR:MainThread} - Fehler in der Ladepunkt-Klasse von 27
Traceback (most recent call last):
  File "/var/www/html/openWB/packages/control/chargepoint/chargepoint.py", line 441, in check_phase_switch_completed
    check_timestamp(self.data.control_parameter.timestamp_charge_start,
  File "/var/www/html/openWB/packages/helpermodules/timecheck.py", line 225, in check_timestamp
    if (create_timestamp() - duration) > timestamp:
TypeError: '>' not supported between instances of 'float' and 'NoneType'`

bin auf dem aktuellen master https://github.com/openWB/core/commit/7799925e9622d2e9bb780bcd1537013946ff47a2

MartinRinas, 2024-12-20:
die relevante Zeile wurde in diesem PR geändert: https://github.com/openWB/core/pull/1974

LKuemmel, 2025-01-02:
Problem ist im verlinkten PR gelöst.

closed 2025-01-02 by LKuemmel