openWB/core #2098: Samba Datensicherung erzeugt leere Datei auf NAS

dl2akt, 2024-12-23:
Seit geraumer Zeit erzeugt die Samba Datensicherung 0 Byte Files auf meinem NAS, das ich zur Datensicherung verwende.
Die Datensicherung lief anfangs ohne Probleme, aber seit einigen Monaten werden nur noch 0 Bytes erzeugt.
Weder auf dem NAS, noch auf der openWB kommen Fehlermeldungen.
Da nun nach etlichen Updates auf die aktuelle Software das Problem immer noch da ist, schreibe ich diese Anfrage.
Die installierte Software-Version ist 2024-12-05 09:57:00 +0100 [d982f63c5].
Was kann ich tun?

Saharel001, 2025-01-17:
Hi, ich hab das SMB Modul ursprünglich entwickelt, schaue mir das am Wochenende mal an. Welches NAS ist in Benutzung mit welchem SMB Protokoll (Version)?

dl2akt, 2025-01-18:
Hi,

vielen Dank für die Antwort. Ich benutze eine Synology DS720+ mit der Betriebssystemversion DSM 7.2.2.

Mindest SMB Protokoll ist 2, Maximales SMB Protokoll ist 3.

 

Das Backup lief mal wie beschrieben eine ganze Weile ohne Probleme. An den SMB Einstellungen hatte ich nie etwas verändert.

Natürlich gab es etliche NAS Updates und auch etliche openWB Updates. Ich kann nicht mehr nachvollziehen, ab wann genau es nicht mehr funktionierte.

 

Bis heute wird jede Nacht eine leere Datei (0 Bytes) auf den NAS Server geschrieben. Zumindest redet openWB mit dem NAS und speichert die Leerdatei auch im angegebenen Ordner.

 

Bei openWB Support hatte ich das Root Passwort abgefragt, aber ich bekam nur als Antwort, dass ich dazu eine neues Rapi Betriebssystem aufsetzen und openWB komplett neu installieren muss, was mir zuviel Aufwand und zu riskant ist, zumal die openWB mit meiner SolarEdge PV Anlage zusammen funktioniert.

 

Wenn ich SSH Zugriff auf die openWB hätte, würde ich selbst ein wenig mit dem SMB Dienst experimentieren, Log Dateilen anschauen.

 

Viele Grüße,

Jens

 

Ich selbst habe ne QNAP im Einsatz mit mit SMB Kompatibilität bis 1.0 und das Backup funktioniert einwandfrei.

@dl2akt <https://github.com/dl2akt>  Wie immer gilt, aktuelles Release einsetzen und damit testen.

benderl, 2025-01-20:
Bei mir läuft es problemlos in Verbindung mit einem Synology NAS mit aktueller 7.2.2 Firmware. Tippe eher auf ein Konfigurationsproblem?

Sieh Dir bitte mal das Backup-Log auf der openWB an. Ist leider noch nicht unter "Fehlersuche" integriert, wie ich gerade gemerkt habe. Du findest es unter `https://192.168.xxx.yyy/openWB/data/log/backup.log`.

dl2akt, 2025-01-20:
Danke für die den Hinweis. Im backup.log finde ich keine Hinweis auf Fehler. Siehe Anhang.

dl2akt, 2025-01-20:
PS: Ich denke, dass dieses Log File nur die Erzeugung des Backup-Files zeigt, aber nicht die Übertragung mit SMB zum NAS. Gibt es auch ein Logfile, das die SMB Übertragung selbst loggt?

closed 2025-01-21 by dl2akt