openWB/core #2101: JSON SoC wird nicht übernommen bei manueller Berechnung

andlem74, 2024-12-27:
Vermutlich wegen des fehlenden Timestamps wird der Wert nicht übernommen, es bleibt der alte (aus der Berechnung resultierende?). Im Log steht:

2024-12-27 10:22:45,091 - {modules.common.store._api:31} - {INFO:store soc_ev1} - Saving CarState(soc=28.0, range=56, soc_timestamp=0)

Ergo: soc_timestamp nicht vorhanden bzw. falsch.

Der Wert von 28% wird unter Status des Fahrzeugs und in der GUI auf der Hauptseite im Color-Theme nicht angezeigt, es bleibt der veraltete Werte (52%). Über MQTT kann ich sehen, dass der gespeicherte SoC-Timestamp des Fahrzeugs einen gültigen Wert > 0 hat. Vermutlich wird der neue Wert deshalb nicht übernommen.

Im Tronity-Modul wird der soc_timestamp gesetzt.

Muss hier eine Timestamp-Generierung im JSON-SoC-Modul hinzugefügt werden, damit es richtig funktioniert? 
Wahlweise auch als JSON-Parse-Ausdruck, damit der Timestamp von der Quelle übernommen wird und ansonsten, wenn diese nicht gegeben ist, als aktueller Zeitpunkt?

andlem74, 2024-12-27:
@MartinRinas , könntest Du da bitte einen Blick drauf werfen?

andlem74, 2024-12-27:
Bei dem Fehler war Lademodus STOP, Fahrzeug gesteckt, SoC-Update per Refresh-Knopf in der GUI.

MartinRinas, 2024-12-27:
Klar schau ich mir gern mal an. Bisher war ein fehlender soc timestamp m.w. kein problem, der würde i.d.r. ja eh nicht beachtet bzw. zentral überschrieben. Es gab kürzlich einen PR um das zu ändern, ggf. spielt das auch eine Rolle. Aber ich kann natürlich einen timestamp mit abrufbar machen und ggf. selbst einen erzeugen. Wobei die Erzeugung m.e.  eher zentral erfolgen sollte sofern keiner übergeben wurde.

andlem74, 2024-12-27:
Ich denke auch, dass es sich lohnen würde, erstmal zu verstehen, warum der SoC zentral nicht (mehr?) übernommen wird. Zumal ich sicher bin, dass es (vor der letzten Release?) auch schon funktioniert hat ohne Zeitstempel.

Du kennst Dich aber besser als ich mit der Verarbeitung der Zeitstempel aus, glaube ich, weil es bei Tronity schon ein Thema war.

MartinRinas, 2024-12-27:
@andlem74  hast Du ein Log von einem nicht erfolgreichen Versuch? Hier sollte der timestamp gesetzt werden sofern keiner (None) gesetzt wird. In Deiner Logzeile ist aber 0 zu sehen, nicht None. Frag' mich gerade wo das herkommt.

https://github.com/openWB/core/blob/a3eaee9f329b462e835e19d602963dda27b8b5cd/packages/modules/common/component_state.py#L141

andlem74, 2024-12-27:
`2024-12-27 10:22:42,883 - {modules.common.configurable_vehicle:56} - {DEBUG:fetch soc_ev1} - Vehicle Instance <class 'modules.vehicles.psacc.config.PSACCVehicleSoc'>
2024-12-27 10:22:42,884 - {modules.common.configurable_vehicle:57} - {DEBUG:fetch soc_ev1} - Calculated SoC-State CalculatedSocState(imported_start=None, manual_soc=None, soc_start=52.0)
2024-12-27 10:22:42,884 - {modules.common.configurable_vehicle:58} - {DEBUG:fetch soc_ev1} - Vehicle Update Data VehicleUpdateData(plug_state=True, charge_state=False, imported=3101969.97, battery_capacity=46000, efficiency=92, soc_from_cp=None, timestamp_soc_from_cp=None)
2024-12-27 10:22:42,884 - {modules.common.configurable_vehicle:59} - {DEBUG:fetch soc_ev1} - General Config GeneralVehicleConfig(use_soc_from_cp=False, request_interval_charging=300, request_interval_not_charging=43200, request_only_plugged=False)
2024-12-27 10:22:42,885 - {modules.common.component_context:26} - {DEBUG:fetch soc_ev1} - Update Komponente ['PSA Car Controller']
2024-12-27 10:22:42,893 - {urllib3.connectionpool:227} - {DEBUG:fetch soc_ev1} - Starting new HTTP connection (1): beaglebone:5000
2024-12-27 10:22:42,893 - {urllib3.connectionpool:227} - {DEBUG:fetch soc_ev1} - Starting new HTTP connection (1): beaglebone:5000
2024-12-27 10:22:44,239 - {urllib3.connectionpool:452} - {DEBUG:fetch soc_ev1} - http://beaglebone:5000 "GET /get_vehicleinfo/VX HTTP/1.1" 200 2131
2024-12-27 10:22:44,239 - {urllib3.connectionpool:452} - {DEBUG:fetch soc_ev1} - http://beaglebone:5000 "GET /get_vehicleinfo/VX HTTP/1.1" 200 2131
2024-12-27 10:22:44,244 - {modules.common.req:32} - {DEBUG:fetch soc_ev1} - Get-Response: {"embedded": null, "links": {"_self": {"deprecation": null, "href": "https://api.groupe-psa.com/connectedcar/v4/user/vehicles//status?profile=endUser", "hreflang": null, "name": null, "profile": null, "templated": null, "title": null, "type": null}, "vehicle": {"deprecation": null, "href": "https://api.groupe-psa.com/connectedcar/v4/user/vehicles/", "hreflang": null, "name": null, "profile": null, "templated": null, "title": null, "type": null}}, "battery": {"current": null, "voltage": 67.0}, "doors_state": null, "energy": [{"updated_at": "2024-12-27 09:08:53+00:00", "autonomy": 56.0, "battery": {"capacity": null, "health": {"capacity": null, "resistance": 100.0}}, "charging": {"charging_mode": "No", "charging_rate": 0, "next_delayed_time": "PT9H1M", "plugged": true, "remaining_time": "PT0S", "status": "Disconnected"}, "consumption": null, "level": 28.0, "residual": null, "type": "Electric"}], "environment": {"created_at": null, "air": {"temp": -1.5}, "luminosity": {"day": false}}, "ignition": {"type": "Stop"}, "kinetic": {"acceleration": 0.0, "moving": false, "pace": null, "speed": 0.0}, "last_position": {"type": "Feature", "geometry": {"coordinates": [9.20179, 48.47619, 400.0], "type": "Point"}, "properties": {"heading": 33.0, "signal_quality": 9.0, "type": "Acquire", "updated_at": "2024-12-27 09:09:06+00:00"}}, "preconditionning": {"air_conditioning": {"failure_cause": null, "status": "Disabled", "updated_at": "2024-12-27 09:04:31+00:00", "created_at": "2024-12-27 09:04:31+00:00"}}, "privacy": {"state": "None"}, "safety": {"belt_warning": null, "e_call_triggering_request": null}, "service": {"type": "Electric", "updated_at": null}, "timed_odometer": {"updated_at": "2024-12-27 09:04:31+00:00", "mileage": 14802.9}}
2024-12-27 10:22:44,246 - {modules.vehicles.json.soc:20} - {DEBUG:fetch soc_ev1} - parse_data: data='{'embedded': None, 'links': {'_self': {'deprecation': None, 'href': 'https://api.groupe-psa.com/connectedcar/v4/user/vehicles/10/status?profile=endUser', 'hreflang': None, 'name': None, 'profile': None, 'templated': None, 'title': None, 'type': None}, 'vehicle': {'deprecation': None, 'href': 'https://api.groupe-psa.com/connectedcar/', 'hreflang': None, 'name': None, 'profile': None, 'templated': None, 'title': None, 'type': None}}, 'battery': {'current': None, 'voltage': 67.0}, 'doors_state': None, 'energy': [{'updated_at': '2024-12-27 09:08:53+00:00', 'autonomy': 56.0, 'battery': {'capacity': None, 'health': {'capacity': None, 'resistance': 100.0}}, 'charging': {'charging_mode': 'No', 'charging_rate': 0, 'next_delayed_time': 'PT9H1M', 'plugged': True, 'remaining_time': 'PT0S', 'status': 'Disconnected'}, 'consumption': None, 'level': 28.0, 'residual': None, 'type': 'Electric'}], 'environment': {'created_at': None, 'air': {'temp': -1.5}, 'luminosity': {'day': False}}, 'ignition': {'type': 'Stop'}, 'kinetic': {'acceleration': 0.0, 'moving': False, 'pace': None, 'speed': 0.0}, 'last_position': {'type': 'Feature', 'geometry': {'coordinates': [9.20179, 48.47619, 400.0], 'type': 'Point'}, 'properties': {'heading': 33.0, 'signal_quality': 9.0, 'type': 'Acquire', 'updated_at': '2024-12-27 09:09:06+00:00'}}, 'preconditionning': {'air_conditioning': {'failure_cause': None, 'status': 'Disabled', 'updated_at': '2024-12-27 09:04:31+00:00', 'created_at': '2024-12-27 09:04:31+00:00'}}, 'privacy': {'state': 'None'}, 'safety': {'belt_warning': None, 'e_call_triggering_request': None}, 'service': {'type': 'Electric', 'updated_at': None}, 'timed_odometer': {'updated_at': '2024-12-27 09:04:31+00:00', 'mileage': 14802.9}}' pattern='.energy[0].level'
2024-12-27 10:22:44,651 - {modules.vehicles.json.soc:20} - {DEBUG:fetch soc_ev1} - parse_data: data='{'embedded': None, 'links': {'_self': {'deprecation': None, 'href': 'https://api.groupe-psa.com/connectedcar/v4//status?profile=endUser',  [... cut ...]  {'updated_at': '2024-12-27 09:04:31+00:00', 'mileage': 14802.9}}' pattern='.energy[0].autonomy'
2024-12-27 10:22:45,039 - {modules.common.configurable_vehicle:67} - {DEBUG:fetch soc_ev1} - Requested start soc from api: 28.0%
2024-12-27 10:22:45,042 - {modules.common.store._api:26} - {DEBUG:fetch soc_ev1} - Raw data CarState(soc=28.0, range=56, soc_timestamp=0)
2024-12-27 10:22:45,091 - {modules.common.store._api:31} - {INFO:store soc_ev1} - Saving CarState(soc=28.0, range=56, soc_timestamp=0)
`

MartinRinas, 2024-12-27:
danke, die letzten drei Zeilen sehen relevant aus, hier wird bei Dir ein soc_timestamp von 0 erzeugt und gespeichert, ich sehe bei mir ein passenden timestamp.

Den Log:
`2024-12-27 10:22:45,039 - {modules.common.configurable_vehicle:67} - {DEBUG:fetch soc_ev1} - Requested start soc from api: 28.0% 
2024-12-27 10:22:45,042 - {modules.common.store._api:26} - {DEBUG:fetch soc_ev1} - Raw data CarState(soc=28.0, range=56, soc_timestamp=0) 
2024-12-27 10:22:45,091 - {modules.common.store._api:31} - {INFO:store soc_ev1} - Saving CarState(soc=28.0, range=56, soc_timestamp=0)`

Mein Log:
`2024-12-27 14:51:02,886 - {modules.common.configurable_vehicle:67} - {DEBUG:fetch soc_ev3} - Requested start soc from api: 42.2%
2024-12-27 14:51:02,886 - {modules.common.store._api:26} - {DEBUG:fetch soc_ev3} - Raw data CarState(soc=42.2, range=0, soc_timestamp=1735311062.886248)
2024-12-27 14:51:02,889 - {modules.common.store._api:31} - {INFO:store soc_ev3} - Saving CarState(soc=42.2, range=0, soc_timestamp=1735311062.886248)` 

Wobei mein Szenario geringfügig anders ist, das hier ist ein Abruf, ohne manuelle Berechnung. Wann genau tritt das Problem bei Dir auf, immer oder z.b. nur nach Beendigung eines Ladevorgangs mit manueller Berechnung des SoC?

andlem74, 2024-12-28:
Szenario war dies: Fahrzeug angesteckt. Ladung beginnt, aber SoC ist veraltet. Lademodus auf Stop und SoC manuell refresht. SoC bleibt veraltet. OpenWB-Neustart, SoC manuell refresht. SoC wird wieder nicht übernommen. Von diesem letzten Refresh ist der Log. 

Danach Wechsel auf Tronity, SoC wird sofort korrekt übernommen.

MartinRinas, 2024-12-28:
Es gelingt mir leider nicht den Fehler nachzustellen, ich bekomme immer wie erwartet einen timestamp vom System zentral erzeugt. Auch wenn ich exakt deinen Schritten folge. Gibt's im main log ggf. Gleichzeitig nicht andere Fehlermeldung? 

Unab davon hab ich einen PR erstellt um den timestamp auch auswerten und übernehmen zu können, das sollte für einige Szenarien dann auch helfen.

andlem74, 2024-12-28:
Danke für den PR.
Sobald ich wieder Zugang zu meiner openWB habe, werde ich versuchen zu debuggen, woher hier die '0' kommt.

andlem74, 2024-12-28:
Beim Durchsehen meiner Logs fällt mir auf, dass auch bei Tronity-SoC CarState-Objekte geloggt werden, deren soc_timestamp=0 ist (und nicht None):
`2024-12-27 10:44:25,571 - {modules.common.configurable_vehicle:56} - {DEBUG:fetch soc_ev1} - Vehicle Instance <class 'modules.vehicles.tronity.config.TronityVehicleSoc'>
2024-12-27 10:44:25,573 - {modules.common.configurable_vehicle:57} - {DEBUG:fetch soc_ev1} - Calculated SoC-State CalculatedSocState(imported_start=None, manual_soc=None, soc_start=52.0)
2024-12-27 10:44:25,574 - {modules.common.configurable_vehicle:58} - {DEBUG:fetch soc_ev1} - Vehicle Update Data VehicleUpdateData(plug_state=True, charge_state=False, imported=3101969.97, battery_capacity=46000, efficiency=92, soc_from_cp=None, timestamp_soc_from_cp=None)
2024-12-27 10:44:25,577 - {modules.common.configurable_vehicle:59} - {DEBUG:fetch soc_ev1} - General Config GeneralVehicleConfig(use_soc_from_cp=False, request_interval_charging=300, request_interval_not_charging=43200, request_only_plugged=False)
2024-12-27 10:44:25,579 - {modules.common.component_context:26} - {DEBUG:fetch soc_ev1} - Update Komponente ['Tronity']
2024-12-27 10:44:25,580 - {modules.vehicles.tronity.api:17} - {DEBUG:fetch soc_ev1} - Fetching Tronity SOC
2024-12-27 10:44:25,583 - {modules.vehicles.tronity.api:26} - {DEBUG:fetch soc_ev1} - No token found
2024-12-27 10:44:25,586 - {modules.vehicles.tronity.api:57} - {DEBUG:fetch soc_ev1} - Requesting new Tronity Access Token
2024-12-27 10:44:25,594 - {urllib3.connectionpool:973} - {DEBUG:fetch soc_ev1} - Starting new HTTPS connection (1): api.tronity.tech:443
2024-12-27 10:44:25,594 - {urllib3.connectionpool:973} - {DEBUG:fetch soc_ev1} - Starting new HTTPS connection (1): api.tronity.tech:443
2024-12-27 10:44:25,962 - {urllib3.connectionpool:452} - {DEBUG:fetch soc_ev1} - https://api.tronity.tech:443 "POST /authentication HTTP/1.1" 201 478
2024-12-27 10:44:25,962 - {urllib3.connectionpool:452} - {DEBUG:fetch soc_ev1} - https://api.tronity.tech:443 "POST /authentication HTTP/1.1" 201 478
2024-12-27 10:44:25,965 - {modules.vehicles.tronity.api:68} - {DEBUG:fetch soc_ev1} - Retrieved new Tronity Access Token: 
2024-12-27 10:44:25,966 - {modules.vehicles.tronity.api:44} - {DEBUG:fetch soc_ev1} - saving new access token
2024-12-27 10:44:26,127 - {urllib3.connectionpool:452} - {DEBUG:fetch soc_ev1} - https://api.tronity.tech:443 "GET /tronity/vehicleslast_record HTTP/1.1" 200 236
2024-12-27 10:44:26,127 - {urllib3.connectionpool:452} - {DEBUG:fetch soc_ev1} - https://api.tronity.tech:443 "GET /tronity/vehicles/last_record HTTP/1.1" 200 236
2024-12-27 10:44:26,130 - {modules.vehicles.tronity.api:99} - {DEBUG:fetch soc_ev1} - Retrieved Tronity data: {'odometer': 14802.9, 'range': 56, 'level': 28, 'charging': 'Charging', 'plugged': True, 'chargerPower': 1.03125, 'chargeRemainingTime': None, 'latitude': 48.47619, 'longitude': 9.20179, 'status': None, 'timestamp': 1735290532000, 'lastUpdate': 1735290532000}
2024-12-27 10:44:26,136 - {modules.common.configurable_vehicle:67} - {DEBUG:fetch soc_ev1} - Requested start soc from api: 28%
2024-12-27 10:44:26,141 - {modules.common.store._api:26} - {DEBUG:fetch soc_ev1} - Raw data CarState(soc=28, range=56, soc_timestamp=1735290532000)
2024-12-27 10:44:26,159 - {modules.common.store._api:31} - {INFO:store soc_ev1} - Saving CarState(soc=28, range=56, soc_timestamp=1735290532000)
2024-12-27 10:49:27,280 - {modules.common.configurable_vehicle:56} - {DEBUG:fetch soc_ev1} - Vehicle Instance <class 'modules.vehicles.tronity.config.TronityVehicleSoc'>
2024-12-27 10:49:27,282 - {modules.common.configurable_vehicle:57} - {DEBUG:fetch soc_ev1} - Calculated SoC-State CalculatedSocState(imported_start=3101969.97, manual_soc=None, soc_start=28)
2024-12-27 10:49:27,283 - {modules.common.configurable_vehicle:58} - {DEBUG:fetch soc_ev1} - Vehicle Update Data VehicleUpdateData(plug_state=True, charge_state=True, imported=3102044.92, battery_capacity=46000, efficiency=92, soc_from_cp=None, timestamp_soc_from_cp=None)
2024-12-27 10:49:27,287 - {modules.common.configurable_vehicle:59} - {DEBUG:fetch soc_ev1} - General Config GeneralVehicleConfig(use_soc_from_cp=False, request_interval_charging=300, request_interval_not_charging=43200, request_only_plugged=False)
2024-12-27 10:49:27,289 - {modules.common.component_context:26} - {DEBUG:fetch soc_ev1} - Update Komponente ['Tronity']
2024-12-27 10:49:27,292 - {modules.vehicles.common.calc_soc.calc_soc:17} - {DEBUG:fetch soc_ev1} - SoC-Gain: ((0.0749499999997206kWh charged * 92% efficiency) / 46.0kWh battery-size) * 100 = 0.1498999999994412%
2024-12-27 10:49:27,294 - {modules.vehicles.common.calc_soc.calc_soc:21} - {DEBUG:fetch soc_ev1} - 28% + 0.06895399999974296kWh = 28.14989999999944%
2024-12-27 10:49:27,295 - {modules.common.configurable_vehicle:67} - {DEBUG:fetch soc_ev1} - Requested start soc from calculation: 28.14989999999944%
2024-12-27 10:49:27,297 - {modules.common.store._api:26} - {DEBUG:fetch soc_ev1} - Raw data CarState(soc=28.14989999999944, range=None, soc_timestamp=0)
2024-12-27 10:49:27,401 - {modules.common.store._api:31} - {INFO:store soc_ev1} - Saving CarState(soc=28.14989999999944, range=None, soc_timestamp=0)`

andlem74, 2024-12-28:
Bei mir läuft die 2.1.6-Stable-Release, abgesehen vom neuen JSON-SoC-Modul. Ein Vergleich mit der Latest zeigt mir, dass die CarState-Klasse in der 2.1.6 den soc_timestamp mit default=0 initialisiert und auch keinen zentralen Timestamp vergibt. Das erklärt, warum ich '0' und nicht 'None' oder einen automatisch generierten Timestamp sehe. Das JSON-SoC-Modul müsste aber m. E. auch mit der 2.1.6 funktionieren, falls diese fehlerfrei ist. Ich würde dann aber diesen Issue hier schließen, weil ja keine Fehler in veraltetem Code mehr verfolgt werden sollen. Ich hoffe, dass die 2.1.7-Release den Fehler dann nicht mehr zeigt.

closed 2024-12-28 by andlem74