openWB/core #2120: JSON SoC versucht, Timestamp-String nach float zu wandeln

andlem74, 2025-01-05:
Die Funktion parse_data in soc.py wandelt den Rückgabewert immer in einen float. Dies schlägt fehl, wenn der Timestamp geparset wird, weil es sich dabei um einen String handelt.
@MartinRinas Kannst Du mal danach schauen?

MartinRinas, 2025-01-05:
Klar kann ich gern machen. Hast du ggf. logs oder Beispieldaten welche Fehlschlagen?

andlem74, 2025-01-05:
Mit einer Zahl geht es:

parse_data({'energy': [{'updated_at': '2025-01-04 10:00:13+00:00', 'autonomy': 214.0, 'battery': None, 'charging': {'charging_mode': 'No', 'charging_rate': 0, 'next_delayed_time': 'PT9H1M', 'plugged': True, 'remaining_time': 'PT0S', 'status': 'Disconnected'}, 'consumption': None, 'level': 69.0, 'residual': None, 'type': 'Electric'}]}, '.energy[0].level')

69.0

Mit dem Timestamp-String nicht:

parse_data({'energy': [{'updated_at': '2025-01-04 10:00:13+00:00', 'autonomy': 214.0, 'battery': None, 'charging': {'charging_mode': 'No', 'charging_rate': 0, 'next_delayed_time': 'PT9H1M', 'plugged': True, 'remaining_time': 'PT0S', 'status': 'Disconnected'}, 'consumption': None, 'level': 69.0, 'residual': None, 'type': 'Electric'}]}, '.energy[0].updated_at')
Traceback (most recent call last):
  File "<python-input-2>", line 1, in <module>
    parse_data({'energy': [{'updated_at': '2025-01-04 10:00:13+00:00', 'autonomy': 214.0, 'battery': None, 'charging': {'charging_mode': 'No', 'charging_rate': 0, 'next_delayed_time': 'PT9H1M', 'plugged': True, 'remaining_time': 'PT0S', 'status': 'Disconnected'}, 'consumption': None, 'level': 69.0, 'residual': None, 'type': 'Electric'}]}, '.energy[0].updated_at')
    ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "parse_data_test.py", line 20, in parse_data
    return float(result)
ValueError: could not convert string to float: '2025-01-04 10:00:13+00:00'

MartinRinas, 2025-01-05:
kannst Du diese Änderungen testen? https://github.com/openWB/core/pull/2118

andlem74, 2025-01-05:
Vielen dank. Fix funktioniert bei mir.

closed 2025-01-05 by andlem74