openWB/core #2136: Display nach update auf 2.6.1 ohne Frontend

0xdefec71f, 2025-01-10:
Servus!

Bei mir läuft openWB auf einem Raspi 3+ mit normalem 7" DSI display, welches eigentlich identisch mit dem der originalen openWB sein sollte. Doch seit dem update auf 2.6.1 wird nach dem reboot nur noch das CLI angezeigt:

[image]

Sowohl das web-frontend und web-Backend lassen sich problemlos im Netzwerk aufrufen. In der Log finde ich Folgendes:

```
detected 'tvservice', query for connected displays
tvservice is not supported when using the vc4-kms-v3d driver.
Similar features are available with standard linux tools
such as kmsprint from kms++-utils

no display detected
displayDetected: false
display setup...
```

In eurer changelog zu V2.6.1 finde ich Folgendes:

> Display-Standby: Der Display-Treiber bietet kein "Tippen zum Aufwecken". Bevor das Display in Standby geht, wird deshalb ein Overlay angezeigt und beim ersten Tipp verschwindet das Overlay, sodass man mit dem ersten Antippen nicht direkt eine Aktion auslöst.

Ich denke, das ist hier bei mir das Problem. Ich kann nur nicht ganz überblicken, was die tatsächliche Änderung ist und wie man das lösen kann. Ich habe bereits `dtoverlay=vc4-fkms-v3d` ausprobiert, da dies angeblich tvservice kompatibel sein sollte. Ohne Erfolg.

Dann habe ich euren zugehörigen code change heraus gesucht: 
[https://github.com/openWB/core/pull/1890/files#diff-6c6ad2100bac1bad54df5c428153d2a893962ed5c0e107b22b1d255c713c9571](https://github.com/openWB/core/pull/1890/files#diff-6c6ad2100bac1bad54df5c428153d2a893962ed5c0e107b22b1d255c713c9571)

Mit `sudo mv /usr/bin/tvservice /usr/bin/tvservice_bak` habe ich dann das tvservice binary versteckt, damit ich beim booten in den else zweig rutsche und vielleicht weiter komme. Die Fehlermeldungen im Log sind nun weg, doch das display zeigt weiterhin nur das cli. Die weitere Log sieht für mich unauffällig aus (komplette Log im Anhang):
```
...
'tvservice' not found, assuming a display is present
displayDetected: true
display setup...
autologin configured
...
```
[Main-Log.txt](https://github.com/user-attachments/files/18381954/Main-Log.txt)

Ich bin nun ratlos... In der Vergangenheit gab es ja schon ein paar Mal Kompatibilitätsprobleme mit dem Display bei V2.x. Ich kann verstehen, dass sich die Entwickler bei openWB auf Ihre eigene Hardware fokussieren und diese auch primär supporten. Ein 'open' Projekt lebt aber auch von den Beiträgen der externen Entwickler und Tester. Mit proprietärer Hardware ist das nicht mehr möglich.

Was ist an meinem Display anders? Ist es möglich hier wieder eine Kompatibilität zu schaffen? Kann mir jemand einen Hinweis geben, was ich noch probieren kann? 

Danke und VG

benderl, 2025-01-14:
Für Eigeninstallationen (Software) können wir leider keinen Support bieten, da es zu viele Möglichkeiten der Konfiguration eines Linux Systems gibt. Bitte unser vorbereitetes Image nutzen. Ich gehe stark davon aus, dass damit auch Dein Display funktioniert, wenn es denn in den Einstellungen aktiviert wurde.

closed 2025-01-14 by benderl

0xdefec71f, 2025-01-16:
> Bitte unser vorbereitetes Image nutzen. Ich gehe stark davon aus, dass damit auch Dein Display funktioniert, wenn es denn in den Einstellungen aktiviert wurde.

@benderl Selbstverständlich ist das Display aktiviert. Diese Aussage überrascht mich jedoch, denn es ist noch gar nicht so lange her, dass mir in einem anderen issue dazu geraten wurde, bei Displayproblemen eben nicht das fertige image zu nehmen. Denn bei diesem funktionieren erst Recht keine anderen Displays.

So wie ich das sehe, habe ich euch darauf aufmerksam gemacht, dass der oben genannte Else-Zweig ganz offensichtlich nicht den Fall abfängt, für den er gedacht war.

Und genau darum geht es hier doch in einem "open" Projekt, welches ihr sogar im Namen tragt. Ihr profitiert von den Meldungen und den Beiträgen aller user. Damit könnt ihr euer Produkt verbessern, welches ihr gewinnbringend verkauft. Zumindest dachte ich das bisher. Ich lasse mich da gerne korrigieren. Es wäre wirklich hilfreich das vielleicht mal irgendwo prominent hinzuschreiben, was man erwarten kann (und was nicht), wenn man sich eurem Projekt unter GPLv3 Lizenz anschließt. Es ist ja schließlich nicht so, als wenn ich hier den selben Support verlangen würde, wie der Käufer einer über 2.000 Euro teuren Wallbox.

Ich versteh es einfach nicht. Es wird aus allem ein Geheimnis gemacht (z.B. Phasenumschaltung). Schaltpläne und Infos zur Hardware darf man im Forum nicht austauschen. Die Kernkomponenten (z.B. Raspi Shield) werden im Shop nicht angeboten. Und wenn man es dann doch schafft sich eine OpenWB zu bauen, einen Bug findet, der euch irgendwann auf die Füße fallen könnte und diesen sogar haarklein aufbereitet und isoliert, dann wird man mit Ignoranz abgestraft. 

Bitte erklärt es mir und den anderen, die auch immer häufiger das "open" in Frage stellen: Warum "open"? Warum dieses Verhalten?

benderl, 2025-01-20:
> [@benderl](https://github.com/benderl) Selbstverständlich ist das Display aktiviert. Diese Aussage überrascht mich jedoch, denn es ist noch gar nicht so lange her, dass mir in einem anderen issue [...]

Das kann ich so nicht bestätigen. Habe z.B. hier ein HDMI-Display mit ganz anderer Auflösung, was ohne Anpassungen funktioniert. Es gibt keine Zeile Code, welche irgendwie aktiv verhindert, dass andere Displays als das "offizielle" 7" Raspi-Display verwendet werden können.

> So wie ich das sehe, habe ich euch darauf aufmerksam gemacht, dass der oben genannte Else-Zweig ganz offensichtlich nicht den Fall abfängt, für den er gedacht war.

Das offizielle Image verwendet keine v3d Overlays, da es damit zu Problemen bei der Drehung des Displays gekommen ist. Vielleicht liegt hier das Problem? Ist aber rein geraten.

> Und genau darum geht es hier doch in einem "open" Projekt, welches ihr sogar im Namen tragt. Ihr profitiert von den Meldungen und den Beiträgen aller user. Damit könnt ihr euer Produkt verbessern, [...]

Du wirfst hier einiges durcheinander. openWB verkauft Wallboxen, die mit dieser open source Software ausgestattet sind. Die Software wird nicht verkauft, sondern durch die verkaufte Hardware subventioniert. Weiterhin bedeutet "open source", dass der Quellcode frei verfügbar ist. Es bedeutet nicht, dass der Autor dieser Software sich um alle möglichen individuellen Einzel- und Sonderfälle kümmern muss. Wenn Du ein Problem bei der Installation hast, was durch eine Anpassung im Code behoben werden kann, stell bitte einen PR, den wir uns dann genauer ansehen können.

> Ich versteh es einfach nicht. Es wird aus allem ein Geheimnis gemacht (z.B. Phasenumschaltung). Schaltpläne und Infos zur Hardware darf man im Forum nicht austauschen. Die Kernkomponenten (z.B. Raspi Shield) werden im Shop nicht angeboten.

Die Hardware ist nunmal nocht open source, sondern Betriebsgeheimnis. Das mag inzwischen nicht mehr ganmz nachvollziehbar zu sein, da es auch andere Wallboxen mit 1p3p Umschaltung gibt, das ändert aber nichts an dieser Regelung.

> Und wenn man es dann doch schafft sich eine OpenWB zu bauen, einen Bug findet, der euch irgendwann auf die Füße fallen könnte und diesen sogar haarklein aufbereitet und isoliert, dann wird man mit Ignoranz abgestraft.

Bitte erstell doch einen PR, den wir uns dann ansehen und testen können. Mit Ignoranz hat das nichts zu tun. Die Begründung hatte ich bereits in meiner ersten Antwort geschrieben:

> Für Eigeninstallationen (Software) können wir leider keinen Support bieten, da es zu viele Möglichkeiten der Konfiguration eines Linux Systems gibt.

Bitte nutze unser Forum, um Dein Problem zu diskutieren und zu lösen. Dort sind wesentlich mehr kompetente User aktiv als hier auf GitHub. Daher wirst Du dort mit großer Sicherheit schneller zum Ziel kommen. Bitte vergiss aber nicht, genau zu beschreiben, wie Du Dein System installiert und konfiguriert hast.

> Bitte erklärt es mir und den anderen, die auch immer häufiger das "open" in Frage stellen: Warum "open"? Warum dieses Verhalten?

"open" -> "open source Software", nicht Hardware. Habe ich jetzt hoffentlich oft genug erläutert.

0xdefec71f, 2025-01-20:
> Bitte unser vorbereitetes Image nutzen. Ich gehe stark davon aus, dass damit auch Dein Display funktioniert, wenn es denn in den Einstellungen aktiviert wurde.

Deine Worte hier klingen ganz anders: [https://github.com/openWB/core/issues/1504#issuecomment-2025802238](https://github.com/openWB/core/issues/1504#issuecomment-2025802238)
=> "_Du verwendest vermutlich unser bereitgestelltes Image. Das ist auf den Einsatz bei den Produkten aus dem Shop von openWB zugeschnitten. Unterstützung für andere Hardware können wir dort nicht einbauen._"

> Die Hardware ist nunmal nocht open source, sondern Betriebsgeheimnis. Das mag inzwischen nicht mehr ganmz nachvollziehbar zu sein, da es auch andere Wallboxen mit 1p3p Umschaltung gibt, das ändert aber nichts an dieser Regelung.

Betriebsgeheimnis?? Ihr habt ein paar Relais an GPIOs eines RasPi gehängt. Selbst die Kernkomponente, die EVSE, kommt von evracing.cz und wurde nicht von euch entwickelt. Für Modbus wird ein USB-Adapter verwendet, Relais und Stromzähler sind Standard Hutschienen-Komponenten. Wo ist hier die geheime Raketentechnik, welche schützenswert geschweige denn schützensfähig wäre?

> Bitte nutze unser Forum, um Dein Problem zu diskutieren und zu lösen. Dort sind wesentlich mehr kompetente User aktiv als hier auf GitHub. 

Das hatte ich in der Vergangenheit getan und sofort einen auf den Deckel bekommen, weil ich eben keiner von den zahlenden Kunden bin. Ich hab das mit dem open source schon verstanden, auch im letzten issue. Was ich kritisiere ist, dass eure Software zunehmend auf eure Hardware proprietär zugeschnitten wird.

Lass gut sein, ich ziehe mir gerade das evcc image....

benderl, 2025-01-20:
> Betriebsgeheimnis?? Ihr habt ein paar Relais an GPIOs eines RasPi gehängt. Selbst die Kernkomponente, die EVSE, kommt von evracing.cz und wurde nicht von euch entwickelt. Für Modbus wird ein [...]

Wie bereits geschrieben, war diese "Raketentechnik" einmal Alleinstellungsmerkmal von openWB. Nur weil sich das inzwischen geändert hat, heißt das noch lange nicht, dass die von openWB entwickelte Hardware (ja, da ist noch Hardware, die selbst entwickelt wurde und kein Standardbauteil ist) jetzt komplett dokumentiert werden muss. Ob das jetzt schützenswert ist oder nicht, kann gerne jeder für sich entscheiden. openWB sieht das noch als schützenswert an.

> Das hatte ich in der Vergangenheit getan und sofort einen auf den Deckel bekommen, weil ich eben keiner von den zahlenden Kunden bin. Ich hab das mit dem open source schon verstanden, auch im [...]

Von irgendwas muss die Entwicklung der Software finanziert werden. Bei openWB geschieht das durch den Verkauf der Wallboxen. Wir bemühen uns, die Software so offen (bezüglich eingesetzter Hardware) zu gestalten, wie es machbar ist. Es ist jedoch ein Ding der Unmöglichkeit, sämtliche Fälle abzudecken. Wir hätten Dein Problem sicher im Forum lösen können, wenn Du mehr Infos zu Deiner Installation gemacht hättest.

> Lass gut sein, ich ziehe mir gerade das evcc image....

Dann wünsche ich Dir dabei viel Erfolg. Ist vermutlich der einfachere Weg.