openWB/core #2187: Audi Q6 etron, bei PV laden und Änderung der Stromstärke stoppt der Ladevorgang und beginnt von Neuem

Buzz2912, 2025-02-02:
Hi,

ich habe eine openWB Duo auf Software 2 (2.1.6-Patch.1). Bisher hatte ich nur Tesla Fahrzeuge und keine Probleme beim PV laden.
Der neue Audi Q6 etron verhält sich hier anders. Im PV Modus startet der Ladevorgang bei mir 3-phasig mit 6A. Bei einer Änderung der Stromstärke durch die openWB geht die Ladeleistung auf 0W zurück um dann auf den neuen Wert zu gehen. Das passiert sowohl bei Änderungen nach oben als auch nach unten.
Das zweite komische Phänomen: 
Stelle ich die openWB auf Sofort (3x16A) und den Q6 auf ein Ladeziel zu einer bestimmten Uhrzeit (100% um 7 Uhr), so startet der Q6 (laut BDA zur Batterieschonung) mit einer geringen Stromstärke und regelt diese gegen Ende des Ladevorgangs hoch (ich kann das nicht ändern). Leider versteht sich die openWB nicht mit dem Q6 und die Stromstärke wird nicht nach oben angepasst und das Auto ist zum gewünschten Zeitpunkt nicht voll. Verwende ich Zielladen in der openWB so schaffe ich das gewünschte Ergebnis.

Ich habe noch einen weiteren Test an einem go-e Charger durchgeführt. Hier ist das Verhalten des Q6 einwandfrei. Änderungen der Stromstärke werden ohne Ladeunterbrechung direkt umgesetzt. Auch die Phasenumschaltung akzeptiert er hier problemlos (habe ich nicht in meiner openWB).

Ich habe die openWB im Primary Modus und auch (mit evcc) im Secondary Modus getestet. Das Verhalten mit dem Q6 ist identisch.
Ich trage gerne log Files oder ähnliches bei.

LKuemmel, 2025-02-03:
Für individuelle Probleme bitte das Forum nutzen. Hier sollten nur allgemeingültige Probleme mit der Software als Issue eingereicht werden. Bei einer Kaufbox kannst Du auch die Supportfunktion nutzen.

closed 2025-02-03 by LKuemmel

Buzz2912, 2025-02-03:
Guten Morgen,

Ich sehe das nicht als individuelles Problem. Der Q6 etron und openwb funktionieren nicht zusammen.
Wie allgemeingültig muss es noch werden um hier als Issue zu stehen?

Gruß, S. Schmid